mcneel/rhino3dm #464: rhino3dm 7.14.2 won't install through pip (invalid wheel!)

fstwn, 2022-03-23:
Dear all,
we had this bug alredy some weeks ago. `rhino3dm` 7.14.2 won't install on win through pip:
```
(base) > pip install rhino3dm
Collecting rhino3dm
  Downloading rhino3dm-7.14.2-cp37-cp37m-win_amd64.whl (2.0 MB)
     ---------------------------------------- 2.0/2.0 MB 11.5 MB/s eta 0:00:00
ERROR: rhino3dm has an invalid wheel, rhino3dm has an invalid wheel, .dist-info directory not found
```

Thanks for taking a look, it seems to me this is related with the current PyPi release workflow? Maybe something in there needs changing...

fstwn, 2022-03-23:
Alright, thank you for clarifying! I don't wanted to bother but raise awareness since I noticed while trying to update my research environments this morning :)
Best wishes, Max

sbaer, 2022-03-24:
Hopefully this is fixed with 7.15 release. We still need to fix our github actions for release.

fraguada, 2023-03-22:
This should be fixed with the 8.0.0 release

closed 2023-03-22 by fraguada