mcneel/rhino3dm #467: GetMesh from face return null in Rhino3dm.js

leifriksheim, 2022-04-19:
Hi! I'm working with the rhino3dm.js library, trying to convert a Brep to a mesh so I can view it in Three.js.

I was looking at the implementation of the 3DMLoader and tried to use the same technique as referenced here:
https://github.com/mrdoob/three.js/blob/f0e2b3453f1412b53389beb04add414e3a80023c/examples/jsm/loaders/3DMLoader.js#L1196.

In my code I seem to only get null back from each face of the Brep. Do I need RhinoCompute for this function call?

jesterKing, 2022-04-19:
The rhino3dm.js itself does not contain meshing code. Instead you need to save your 3dm file with one viewport in shader or rendered mode (and all the geometry you are interested in visible) so that the library can read the mesh data generated by Rhino prior to saving.

sbaer, 2022-04-19:
If the mesh is not embedded in the file using the technique described above, then yes you would need to call Rhino.Compute to get meshes from Breps

closed 2022-04-19 by sbaer

sbaer, 2022-04-19:
The forum is best for questions like this.

You would need to create a Mesh in rhino3dm.js, then add 8 vertices and 6 faces to represent a bounding box. This mesh could then be converted into a threejs compatible mesh