mcneel/rhino3dm #470: Missing and wrong Typescript definitions

jesterKing, 2022-04-22:
https://discourse.mcneel.com/t/rhino3dm-js-issues-in-the-typescript-declaration-file/141504

* BasepointZero -> BasePointZero
* double missing
* byte missing

Relates to #433

opdich, 2022-08-11:
I am having the same issue. Additionally, `rhino3dm.d.ts` is not included in the most recent version, 7.15.0

ghaaflan, 2023-01-05:
you need to add a declaration file to convert js modules to typescript.
A declaration file (with the .d.ts extension) is a TypeScript file that contains type declarations for a module, library, or API. These declarations provide type information for JavaScript code, allowing the TypeScript compiler to type-check and provide type-safe access to the code.

`
declare module 'rhino3dm' {
  // Declare the types and interfaces for the `rhino3dm` module here.
  // For example:
  export interface Point {
    x: number;
    y: number;
    z: number;
  }
}
`

ghaaflan, 2023-01-05:
you need to add a declaration file to convert js modules to typescript.
A declaration file (with the .d.ts extension) is a TypeScript file that contains type declarations for a module, library, or API. These declarations provide type information for JavaScript code, allowing the TypeScript compiler to type-check and provide type-safe access to the code.

```
declare module 'rhino3dm' 
 // Declare the types and interfaces for therhino3dmmodule here. 

// For example: 
export interface Point { x: number; y: number; z: number; } 

```

fraguada, 2023-02-06:
Dusting off this issue.

- [x] BasepointZero -> BasePointZero - the js bindings were using Base**P**ointZero (p capitalized) where in the rest of the codebase it is `BasepointZero`.
- [ ] double missing - double doesn't exist in js, and the real issue is that we are not returning the appropriate type for an internal class that we use to create tuples. In the particular case where the ts defs are wrong, we should probably be returning an array or an object.
- [ ] byte missing - here we return a `UInt8Array`, but we are not picking that up in the ts definitions

fraguada, 2023-02-09:
Closing. Tracking additional issues in #431

fraguada, 2024-03-07:
There still seems to be an issue with the d.ts files. Reopening to see if I can get a better system in place.

fraguada, 2024-03-14:
Updated definitions are in the dev branch: https://github.com/mcneel/rhino3dm/blob/dev/src/js/rhino3dm.d.ts