mcneel/rhino3dm #484: C API for easier third-party bindings

NeusFear, 2022-11-11:
Maintaining a C API would make creating a third party bindings. 

It would be unreasonable for us to request things like "add a Java API" or something along those lines, but if you added a C API we could use JNI to bind to the C API ourselves with relative ease. Bindings not limited to just Java obviously, there are many other languages that allow binding to C very simply.

I realize that the community guidelines say that we are supposed to use the YouTrack, but when attempting to do that I get an error saying that there are no more seats available. So I am resorting to here,

sbaer, 2022-11-11:
There actually is a large C API in place as this is what the .NET bindings use for pInvokes. This can be found in the librhino3dm_native directory.

NeusFear, 2022-11-11:
To us in the LWJGL (lightweight java game library) community it seems like the C headers are not all inclusive. I'm not sure if this is because the C# bindings may be generated from the C++ api or something similar, but essentially we would need headers for the full library to be able to bind to it with JNI.

closed 2025-02-28 by fraguada