mcneel/rhino3dm #492: rhino3dm.js keep getting error ERR_INVALID_URL

dgrcode, 2023-01-11:
I was getting this issue with 0.14.0. After upgrading to 7.15.0 I keep getting the same error.

```
TypeError: Failed to parse URL from {project path}/node_modules/rhino3dm/rhino3dm.wasm
    at Object.fetch (node:internal/deps/undici/undici:11118:11)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5) {
  [cause]: TypeError [ERR_INVALID_URL]: Invalid URL
      at new NodeError (node:internal/errors:393:5)
      at URL.onParseError (node:internal/url:565:9)
      at new URL (node:internal/url:645:5)
      at new Request (node:internal/deps/undici/undici:9472:25)
      at Agent.fetch2 (node:internal/deps/undici/undici:10291:25)
      at Object.fetch (node:internal/deps/undici/undici:11116:28)
      at fetch (node:internal/process/pre_execution:216:25)
      at instantiateAsync ({project path}/node_modules/rhino3dm/rhino3dm.js:9:16039)
      at createWasm ({project path}/node_modules/rhino3dm/rhino3dm.js:9:16646)
      at {project path}/node_modules/rhino3dm/rhino3dm.js:9:131229 {
    input: '{project path}/node_modules/rhino3dm/rhino3dm.wasm',
    code: 'ERR_INVALID_URL'
  }
}
```
I'm not sure what's going on, because the file at {project path}/node_modules/rhino3dm/rhino3dm.wasm is there. I can do `cat {project path}/node_modules/rhino3dm/rhino3dm.wasm` and see a bunch of binary output.

I think I might have updated node to v18.12.1 before I started seeing this issue. Could this be a node 18 compatibility issue?

I do get this warning when starting the server:
```
(node:71881) ExperimentalWarning: The Fetch API is an experimental feature. This feature could change at any time
```

dgrcode, 2023-01-11:
I can confirm I don't get this issue with node v16.19.0. We'd have to confirm this is a node 18 incompatibility issue. This would be a problem given node's releases.

From node's release schedule ([link](https://github.com/nodejs/release#release-schedule)), here's the versions lifetime
[image]

v16 entered maintenance status about 3 months ago, on 2022-10-18. End of life will be 2023-09-11.

dgrcode, 2023-01-11:
cross-posted to McNeel YouTrack Application as per CONTRIBUTING.md.

Link: https://mcneel.myjetbrains.com/youtrack/issue/BO-3497/rhino3dm.js-with-Node-18-throws-error-ERRINVALIDURL

sbaer, 2023-01-14:
What does the code that loads rhino3dm in your node server look like?

dgrcode, 2023-01-16:
This is the code that loads rhino3dm

```js
const rhino3dm = require('rhino3dm')

console.log('rhino3dm: initializing')
/**
 * @type Promise<rhino3dm.RhinoModule>
 */
const rhinoPromise = rhino3dm()
  .then((rhino) => {
    console.log('rhino3dm: loaded')
    return rhino
  })
  .catch((err) => {
    console.log('rhino3dm: error', err)
  })

module.exports = rhinoPromise
```

Then, I use it like this:
```js
const rhinoPromise = require('./server/services/rhino3dm')

const rhino = await rhinoPromise
```

But I'm confident that's not the issue, because that exact code works with node v16. It has only stopped working when I updated node to v18

dgrcode, 2023-01-24:
Hi @sbaer any news on this? Can we confirm rhino3dm.js is not compatible with Node v18?

fraguada, 2023-01-24:
No web assembly compiled code is compatible with Node v18.1.0 and above. See: https://github.com/emscripten-core/emscripten/issues/16913

Seems there was a patch applied. We need to see how to build rhino3dm with an updated version of emscripten to get the benefit of the patch. 

In the meantime you can run your scripts like so:
`node --no-experimental-fetch index.js`

dgrcode, 2023-01-24:
Awesome, thanks for the details and workaround 🙌

dgrcode, 2023-02-28:
I can confirm the files in the link work as expected. Thanks for fixing this one. Is there an estimate update release date?

fraguada, 2023-02-28:
Thanks for testing!
We have most everything in place for a release except for testing. Hopefully we can wrap it up in a week or so.

fraguada, 2023-03-13:
@dgrcode Just published rhino3dm.js 8.0.0. https://www.npmjs.com/package/rhino3dm

closed 2023-03-13 by fraguada

dgrcode, 2023-03-14:
Thanks for the heads up! I just upgraded to 8.0.0 and it's working great using node v18

fraguada, 2023-03-14:
Thanks for reporting @dgrcode!

dgrcode, 2023-04-21:
Quick question. I just created a 3dm file with the rhino3dm.js library, and it gave me an error when trying to open it with Rhino. It said something like the file had been created with Rhino 8 and it couldn't be opened with Rhino 7. I haven't seen any Rhino 8 available yet.

Is that expected?

fraguada, 2023-04-30:
answered in #521