mcneel/rhino3dm #493: rhino3dm.d.ts file is missing in published npm package 7.15.0

Jauminha, 2023-02-16:
Hi there @fraguada ,

I was installing the latest version into my TS project and I saw that the `d.ts` file is missing from `7.15`
You can check it by yourself and compare these 2:
https://www.npmjs.com/package/rhino3dm/v/7.15.0?activeTab=explore -> 👎 
https://www.npmjs.com/package/rhino3dm/v/7.14.0?activeTab=explore -> 👍 

You should republish with the right files, there also appears to be a weird `RHINO3DM.JS.md` which seems just a copy of the regular `README.md` file.

I've already seen https://github.com/mcneel/rhino3dm/issues/470 and I'm using the d.ts file from there, but this is just a temporary solution, good enough for now while developing, but the moment this will need to hit production, there is a pipeline that will most defenitely need to have the file there. I'd rather not have to use the `7.14`.

Thanks for time.

fraguada, 2023-02-16:
At the moment we are working on an 8.0.0 release with the fixes discussed in #470.

closed 2023-02-16 by fraguada

fraguada, 2023-02-16:
We're tracking other issues with the d.ts in #431.
If you have others that are not covered by that, feel free to open another issue with the specific methods that aren't working for you.

Jauminha, 2023-02-16:
> We are not catching cases where there is a specific method for js that does not exists for dotnet and python. I'm looking into fixing this.
It is hard to tell what is covered in here and what isn't.