mcneel/rhino3dm #513: DllNotFoundException for 'librhino3dm_native' on linux for versions above 7.6.0

tkahng, 2023-03-29:
```
System.DllNotFoundException: Unable to load shared library 'librhino3dm_native' or one of its dependencies.
```

This error occurs when running on linux-x64 machines with .NET6, for any versions higher than 7.6.0.

 I would really like to upgrade, since there are newly added apis i would like to use, such as Curve.JoinCurves, and would like to move to .NET7.
 
 This issue has been around for a while, but there seems to be no updates on it.

fraguada, 2023-03-29:
Thanks for reporting. I will try to test this on a linux-x64 system and let you know what I find.

tkahng, 2023-03-29:
```
Hello, World!
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile
Unhandled exception. System.DllNotFoundException: Unable to load shared library 'librhino3dm_native' or one of its dependencies. In order to help diagnose loading problems, consider using a tool like strace. If you're using glibc, consider setting the LD_DEBUG environment variable: 
/lib64/libm.so.6: version `GLIBC_2.27' not found (required by /home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/runtimes/linux-x64/native/librhino3dm_native.so)
/usr/share/dotnet/shared/Microsoft.NETCore.App/7.0.2/librhino3dm_native.so: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/librhino3dm_native.so: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/runtimes/linux-x64/native/liblibrhino3dm_native.so: cannot open shared object file: No such file or directory
/usr/share/dotnet/shared/Microsoft.NETCore.App/7.0.2/liblibrhino3dm_native.so: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/liblibrhino3dm_native.so: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/runtimes/linux-x64/native/librhino3dm_native: cannot open shared object file: No such file or directory
/usr/share/dotnet/shared/Microsoft.NETCore.App/7.0.2/librhino3dm_native: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/librhino3dm_native: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/runtimes/linux-x64/native/liblibrhino3dm_native: cannot open shared object file: No such file or directory
/usr/share/dotnet/shared/Microsoft.NETCore.App/7.0.2/liblibrhino3dm_native: cannot open shared object file: No such file or directory
/home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/bin/Debug/net7.0/liblibrhino3dm_native: cannot open shared object file: No such file or directory

   at UnsafeNativeMethods.ONX_Model_New()
   at Rhino.FileIO.File3dm..ctor()
   at Program.<Main>$(String[] args) in /home/ec2-user/Github/rhino-developer-samples/rhino3dm/cs/SampleCSReadFile/Program.cs:line 8
   ```
   
   @fraguada  still getting this error.

tkahng, 2023-03-30:
Okay so it works on al 2023 machines with dotnet 6 and 7. However for lambdas net6 runtime 8 beta gives this:

[image]

And for using the custom runtime turns out to not play with other libraries. 

It would be nice if we could at least know what caused it. Seems like at least til 7.15 was not targeting 7, and the issue was for anything above 7,6 so and it seems like there wasnt much public discussion on what change caused it.

fraguada, 2023-03-31:
I just was able to test this on an aws ec2 linux-x64 instance. I provisioned, connected, installed git, node, dotnet 7 runtime and sdk, pulled the developer samples onto the machine and ran `dotnet run`.
As you've already confirmed, it works.

Next week I will check on aws lambdas.

fraguada, 2023-04-05:
Hello! Today I finally got all the aws going and started a new dotnet lambda project. I followed this guide: https://docs.aws.amazon.com/lambda/latest/dg/csharp-package-cli.html

First I made an aws lambda net6.0:

[image]

The function cs looks like this:

```csharp
using Amazon.Lambda.Core;
using Rhino.Geometry;

// Assembly attribute to enable the Lambda function's JSON input to be converted into a .NET class.
[assembly: LambdaSerializer(typeof(Amazon.Lambda.Serialization.SystemTextJson.DefaultLambdaJsonSerializer))]

namespace my_rhino_function;

public class Function
{
    
    /// <summary>
    /// A simple function that takes a string number and uses that as a radius for a sphere.
    /// </summary>
    /// <param name="input"></param>
    /// <param name="context"></param>
    /// <returns></returns>
    public string FunctionHandler(string input, ILambdaContext context)
    {
        var sphere = new Sphere(Point3d.Origin, double.Parse(input));

        return "Radius: " + input + " Diameter: " + sphere.Diameter.ToString();
    }
}

```

invoking the function works fine.

fraguada, 2023-04-05:
Trying to open a file I do get the DllNotFoundException. But I wonder if this is because rhino3dm is for net7.0

fraguada, 2023-04-11:
@tkahng I looked back at how we were building the linux version around 7.6.0 and see that it is when we switched from CircleCI to GitHub Actions, leaving a few of the build conditions behind. It seems we were building a linux version on an Amazon Linux based image, but I am unclear if we actually shipped this version with the nuget package. 

The latest code built with this image seems to work on aws lambda. If you are brave, you can try out [this artifact](https://github.com/mcneel/rhino3dm/suites/12169330056/artifacts/642358334) which is built with the same docker image we used for 7.6.0, only difference is that I've updated to dotnet 7. Still, I've tested this on aws lambda targeting net 6 and net 7 and works in both cases. My testing hasn't been extensive.

tkahng, 2023-04-11:
Thanks for the info. Ill try again for 6, since 7 doesnt seem to work for some of ny dependencies.

tkahng, 2023-04-11:
@fraguada For rhino3dm 8, on dotnet 6, does DllNotFoundException still occur when reading files on dotnet 6?

fraguada, 2023-04-12:
@tkahng I've tested two templates:

- Lambda Empty Function - .NET 6.0
- Lambda Empty Function (.NET 7 Container Image) - .NET 7.0

Both running the same code: 

```csharp
    public string FunctionHandler(string input, ILambdaContext context)
    {
        var file3dm = Rhino.FileIO.File3dm.Read("RectifiedArchimedeanSolids8.3dm");        
        return "Number of objects in file " +  file3dm.Objects.Count.ToString();
    }
```
The csproj file is like this:

```xml
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <TargetFramework>net6.0</TargetFramework>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
    <GenerateRuntimeConfigurationFiles>true</GenerateRuntimeConfigurationFiles>
    <AWSProjectType>Lambda</AWSProjectType>
    <!-- This property makes the build directory similar to a publish directory and helps the AWS .NET Lambda Mock Test Tool find project dependencies. -->
    <CopyLocalLockFileAssemblies>true</CopyLocalLockFileAssemblies>
    <!-- Generate ready to run images during publishing to improve cold start time. -->
    <PublishReadyToRun>true</PublishReadyToRun>
  </PropertyGroup>
  <ItemGroup>
    <PackageReference Include="Amazon.Lambda.Core" Version="2.1.0" />
    <PackageReference Include="Amazon.Lambda.Serialization.SystemTextJson" Version="2.3.1" />
  </ItemGroup>
  <ItemGroup>
    <Reference Include="Rhino3dm">
      <HintPath>Rhino3dm.dll</HintPath>
    </Reference>
  </ItemGroup>
  <Target Name="CopyCustomContent" AfterTargets="AfterBuild">
    <Copy SourceFiles="RectifiedArchimedeanSolids8.3dm" DestinationFolder="$(OutDir)" />
    <Copy SourceFiles="librhino3dm_native.so" DestinationFolder="$(OutDir)" />
  </Target>
  <Target Name="CopyCustomContentOnPublish" AfterTargets="Publish">
    <Copy SourceFiles="RectifiedArchimedeanSolids8.3dm" DestinationFolder="$(PublishDir)" />
    <Copy SourceFiles="librhino3dm_native.so" DestinationFolder="$(PublishDir)" />
  </Target>
</Project>
```

This is mostly because I am testing the library as I build it. Normally you'd have rhino3dm as a Package Reference. 

And the files:
[image]

fraguada, 2023-04-17:
I just published a rhino3dm8.0.0-beta.1 on nuget: https://www.nuget.org/packages/Rhino3dm/8.0.0-beta.1 with the fixes discussed here.

tkahng, 2023-04-20:
@fraguada im getting this error again with the 8.0.0-beta.1 
[image]
this was run in an al2 ec2 with dotnet 6, also same with lambda dotnet 6

sanchez, 2023-05-17:
I have the same problem on net6 Rhino3dm 7.15, trying to deserialize a Rhino3dm file from RhinoCompute throws an error on Linux App Service on Azure

fraguada, 2023-05-17:
@tkahng did you try dotnet 7?

@sanchez have you tried the latest rhino3dm.net 8.0.0 beta? What is the architecture of the Linux App Service on Azure?

sanchez, 2023-05-17:
The architecture is just the Azure Linux App Service running on tier B1 of the App Service Plan with Zip Deploy, no docker or anything like that

fraguada, 2023-05-18:
@sanchez is it arm64 or x64? Which version of linux?

sanchez, 2023-05-18:
[image]

I found this information and `uname -m` return `x86_64`

fraguada, 2023-05-18:
I think Azure App Service runs on Debian. I've not tested there

fraguada, 2023-11-02:
btw @sanchez I've just tested rhino3dm with the Windows and Linux OS variants of Azure Functions and it works fine. See this example: https://github.com/mcneel/rhino-developer-samples/tree/8/rhino3dm/cs/SampleCSAzureFunction

I'll try next with App Service.

fraguada, 2023-11-06:
@sanchez I've tested rhino3dm 8.0.0-beta3 in an Azure AppService and can get rhino3dm to compute some stuff and render values to a view. I'll mark this as closed now, but write here if you are still having issues.

closed 2023-11-06 by fraguada

sanchez, 2023-11-07:
Cheers @fraguada, unfortunately at this current point in time I don't have capacity to verify on my side, however if I encounter this issue I will reopen the ticket.

Thanks

fraguada, 2023-11-07:
btw, here is an Azure App Service example: https://github.com/mcneel/rhino-developer-samples/tree/8/rhino3dm/cs/SampleCSAzureAppService

Apologies for the simplicity!