mcneel/rhino3dm #565: Upload Linux wheels to PyPI

robbievanleeuwen, 2023-09-27:
Hi thanks for taking the time to develop and maintain this awesome project! We use `rhino3dm` over at [sectionproperties](https://github.com/robbievanleeuwen/section-properties) to allow our users to import Rhino `.3dm` files and undertake complex cross-section analysis.

Installing `rhino3dm` is part of our CI (github actions and readthedocs build) and we are currently having issues with installing this package on Linux for python 3.10 and 3.11. I've noticed that there are Windows and Mac wheels for `rhino3dm` on PyPI for all versions of python, however there are only wheels for Linux up to python 3.9.

We were wondering if it would be possible to upload Linux wheels to your PyPI releases for python 3.10 and 3.11 to allow us to efficiently use these versions in our CI without having to rebuild `rhino3dm` with each commit? Also open to any other solutions that may work!

Thanks!

fraguada, 2023-10-04:
@robbievanleeuwen We build a bunch of pythin linux versions: https://github.com/mcneel/rhino3dm/actions/runs/6191077082

I need to remember why we don't push these to pypi. Would including the linux py whls that are in the link above with the pypi package help your process?

robbievanleeuwen, 2023-10-04:
Hi @fraguada, yes this would solve the issue! 

At the moment we don't want to build the `rhino3dm` wheel for our linux tests every time our CI runs as this takes a long time, ideally we would just pull the pre-built wheel from pypi. Further our readthedocs build (which includes some rhino examples) would fail as the entire build time would take greater than the 15 min limit if we also had to build the `rhino3dm` wheel.

fraguada, 2023-10-04:
OK! I will include these in the 8.0.0b3 release. Hopefully I can put that together in the next week or so.

robbievanleeuwen, 2023-10-04:
@fraguada much appreciated!

fraguada, 2023-10-10:
I've added cp310 and cp311 manylinux whl. https://pypi.org/project/rhino3dm/8.0.0b3/
Let me know if this takes care of your issues!

closed 2023-10-10 by fraguada

robbievanleeuwen, 2023-10-11:
@fraguada yes seems to be working fine for python 3.10 & 3.11 on Linux now - thanks for fixing this for us 😄