mcneel/rhino3dm #606: Initialization error for AWS lambda in rhino3dm.js 8.6.0

harrycollin, 2024-04-18:
Hey all,

I upgraded from rhino3dm.js from 8.4.0 to 8.6.0 and get an obscure error when trying to initialize the module.

```
TypeError: Cannot read properties of undefined (reading 'value')
    at Object.toValue (file:///var/task/path/to/function/handle-model-uploaded.mjs:134166:45)
    at __emval_run_destructors (file:///var/task/path/to/function/handle-model-uploaded.mjs:134796:35)
    at wasm://wasm/009bea62:wasm-function[4831]:0x117e94
    at wasm://wasm/009bea62:wasm-function[3552]:0xafbeb
    at callRuntimeCallbacks (file:///var/task/path/to/function/handle-model-uploaded.mjs:130143:30)
    at initRuntime (file:///var/task/path/to/function/handle-model-uploaded.mjs:129951:11)
    at doRun (file:///var/task/path/to/function/handle-model-uploaded.mjs:135222:13)
    at run (file:///var/task/path/to/function/handle-model-uploaded.mjs:135238:13)
    at runCaller (file:///var/task/path/to/function/handle-model-uploaded.mjs:135202:13)
    at removeRunDependency (file:///var/task/path/to/function/handle-model-uploaded.mjs:130004:15)
  ```
  
This doesn't tell me much, I was hoping someone else might have some insight into the issue. I also welcome any advise on how to get more info about this error. Thanks!

Lambda runtime: NodeJS 18.x
Lambda architecture: x86
Lambda memory: 10GB

fraguada, 2024-04-24:
I will try to create a similar environment to see if I get the same issue.

kovacsv, 2024-04-27:
Same issue here: #608. No need for AWS, you can reproduce it with [this fiddle](https://jsfiddle.net/kovacsv/2d7th643).

fraguada, 2024-05-09:
@harrycollin I am preparing an 8.6.1 release for js that should fix this issue. If you can test this package, I would really appreciate it: https://github.com/mcneel/rhino3dm/actions/runs/9017758138/artifacts/1487757185

fraguada, 2024-05-10:
https://www.npmjs.com/package/rhino3dm/v/8.6.1 has been published which should resolve this issue. Please let me know if that is not the case.

closed 2024-05-10 by fraguada

harrycollin, 2024-05-14:
@fraguada looks like it's working now. Thanks!