mcneel/rhino3dm #607: Import error on arm64 architecture (M1)

antalvarenga-ic, 2024-04-23:
Hi,

I am trying to use rhino3dm inside a poetry project, but it's throwing an error:

```
...
File "/Users/aalvarenga/apps/neom/neom-be/.venv/lib/python3.11/site-packages/rhino3dm/__init__.py", line 5, in <module>
    from ._rhino3dm import *
ImportError: dlopen(/Users/aalvarenga/apps/neom/neom-be/.venv/lib/python3.11/site-packages/rhino3dm/_rhino3dm.cpython-311-darwin.so, 0x0002): tried: '/Users/aalvarenga/apps/neom/neom-be/.venv/lib/python3.11/site-packages/rhino3dm/_rhino3dm.cpython-311-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/aalvarenga/apps/neom/neom-be/.venv/lib/python3.11/site-packages/rhino3dm/_rhino3dm.cpython-311-darwin.so' (no such file), '/Users/aalvarenga/apps/neom/neom-be/.venv/lib/python3.11/site-packages/rhino3dm/_rhino3dm.cpython-311-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
```

I installed rhino3dm using poetry:
`poetry add rhino3dm`

`poetry env info` output:

```
Virtualenv
Python:         3.11.7
Implementation: CPython
Path:           /Users/aalvarenga/apps/neom/neom-be/.venv
Executable:     /Users/aalvarenga/apps/neom/neom-be/.venv/bin/python
Valid:          True

System
Platform:   darwin
OS:         posix
Python:     3.11.7
Path:       /Library/Frameworks/Python.framework/Versions/3.11
Executable: /Library/Frameworks/Python.framework/Versions/3.11/bin/python3.11
```

OS is macOS 14.2.1 M1

fraguada, 2024-04-27:
Seeing this issue. Seems we never actually build a multi architecture native library for macos.

fraguada, 2024-04-28:
@antalvarenga-ic I believe I fixed this. I added arm64 architecture to the macos python builds and have tested the cp311 build on my M1. Can you try the artifact here: https://github.com/mcneel/rhino3dm/actions/runs/8865480206/artifacts/1454240476

I installed it like so: `python3 -m pip install rhino3dm-8.6.0-cp311-cp311-macosx_11_0_universal2.whl`

fraguada, 2024-04-28:
Tried a few things. GitHub has arm64 runners so I was able to build on that. I tested on my M1 and it installs and runs. Try it and let me know if it works for you: https://github.com/mcneel/rhino3dm/actions/runs/8869285668/artifacts/1454749544

and install with: pip install rhino3dm-8.6.1-cp311-cp311-macosx_14_0_universal2.whl

closed 2024-04-28 by fraguada

antalvarenga-ic, 2024-04-29:
Sorry for the delay. I just tested it and it works. Thank you.

fraguada, 2024-05-10:
@antalvarenga-ic I just published rhino3dm 8.6.1 which includes macos arm64 wheels