mcneel/rhino3dm #608: rhino3dm.js 8.6.0 can't be initialized

kovacsv, 2024-04-27:
Initialization of version 8.6.0 fails with the following message. Version 8.4.0 works fine. See [this fiddle](https://jsfiddle.net/kovacsv/2d7th643) for an example.
```
Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'value')
    at Object.toValue (rhino3dm.min.js:1:59353)
    at s (rhino3dm.min.js:1:79709)
    at rhino3dm.wasm:0x117e94
    at rhino3dm.wasm:0xafbeb
    at ie (rhino3dm.min.js:1:4539)
    at r (rhino3dm.min.js:1:85790)
    at vt (rhino3dm.min.js:1:86274)
    at e (rhino3dm.min.js:1:86302)
    at Y (rhino3dm.min.js:1:3943)
    at r (rhino3dm.min.js:1:83528)
```

fraguada, 2024-04-27:
thanks, checking it out

xRadne, 2024-04-29:
Had the same issue when upgrading to 8.6.0

fraguada, 2024-05-09:
@kovacsv @xRadne I am preparing an 8.6.1 release for js that should fix this issue. If you can test this package, I would really appreciate it: https://github.com/mcneel/rhino3dm/actions/runs/9017758138/artifacts/1487757185

xRadne, 2024-05-09:
Thank you! I will test it first thing tomorrow morning

xRadne, 2024-05-10:
@fraguada I have now tested 8.6.1 and can confirm that this have solved the initialization error! 

Thank you for fixing this :)

fraguada, 2024-05-10:
Thank you for testing! I will prepare a release on npm.

fraguada, 2024-05-10:
https://www.npmjs.com/package/rhino3dm/v/8.6.1 has been published which should resolve this issue. Please let me know if that is not the case.

closed 2024-05-10 by fraguada