mcneel/rhino3dm #610: Inconsistant PolylineCurve.TryGetPolyline() method

xarthurx, 2024-04-29:
For the python lib, when trying to convert from `PolylineCurve` to `Polyline`, the actual function usage is
```python
ply = plyCrv.TryGetPolyline()
```
However, the linting and code are all `TryGetPolyline(polyline: Polyline)` which seems to be inherited from .NET method, but not working in Python.
[image]
[image]

fraguada, 2024-05-10:
I can imagine that the stubs are very out of date. That being said, there does exist a Curve.TryGetPolyline(out Polyline polyline) method in dotnet. It returns a bool on whether the attempt was successful. In python, the result is the curve or None: https://github.com/mcneel/rhino3dm/blob/main/src/bindings/bnd_curve.cpp#L61

I will see about getting the stubs updated to reflect this.

xarthurx, 2024-05-10:
> I can imagine that the stubs are very out of date. That being said, there does exist a Curve.TryGetPolyline(out Polyline polyline) method in dotnet. It returns a bool on whether the attempt was [...]

There is.

The polyline is obtained with 'out'.
The way dotNet and python works very differently.

closed 2025-02-28 by fraguada