mcneel/rhino3dm #612: [ERROR] Could not resolve "ws"

hopezh, 2024-05-05:
I got the following error while running the example code [rhino3dm.js](https://github.com/mcneel/rhino3dm/tree/main#rhino3dmjs-javascript-and-nodejs). 

I don't know how to **`...mark the path "ws" as external to exclude it from the bundle`**, as suggested in the error message. 

May I ask if I have missed something important here?  

```shell
npm run dev

> threejs_starter_v01@0.0.0 dev
> vite

  VITE v5.2.8  ready in 169 ms

  ➜  Local:   http://localhost:5173/
  ➜  Network: http://192.168.138.1:5173/
  ➜  Network: http://192.168.70.1:5173/
  ➜  Network: http://192.168.12.121:5173/
  ➜  press h + enter to show help

✘ [ERROR] Could not resolve "ws"

    node_modules/rhino3dm/rhino3dm.js:9:64363:
      9 │ ...ocketServer=require("ws").Server;var host=so...
        ╵                        ~~~~

  You can mark the path "ws" as external to exclude it from
  the bundle, which will remove this error and leave the
  unresolved path in the bundle. You can also surround this
  "require" call with a try/catch block to handle this
  failure at run-time instead of bundle-time.

6:16:26 pm [vite] error while updating dependencies:
Error: Build failed with 1 error:
node_modules/rhino3dm/rhino3dm.js:9:64363: ERROR: Could not resolve "ws"
    at failureErrorWithLog (C:\Users\aaa\Downloads\code\threejs_starter_v01\node_modules\esbuild\lib\main.js:1651:15)
    at C:\Users\aaa\Downloads\code\threejs_starter_v01\node_modules\esbuild\lib\main.js:1059:25
    at C:\Users\aaa\Downloads\code\threejs_starter_v01\node_modules\esbuild\lib\main.js:1527:9
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
```

The output of my `npm list`:
```shell
+-- rhino3dm@8.6.0
+-- three@0.163.0
`-- vite@5.2.8
```

fraguada, 2024-05-09:
Please try with rhino3dm@8.4.0 as there is an issue with 8.6.0

fraguada, 2024-05-09:
I am preparing an 8.6.1 release for js. If you can test this package, I would really appreciate it: https://github.com/mcneel/rhino3dm/actions/runs/9017758138/artifacts/1487757185

fraguada, 2024-05-10:
https://www.npmjs.com/package/rhino3dm/v/8.6.1 has been published which should resolve this issue. Please let me know if that is not the case.

hopezh, 2024-05-10:
Thanks, @fraguada. 

I still got the following error while running the example code on [the latest npm repo of rhino3dm](https://www.npmjs.com/package/rhino3dm/v/8.6.1), and the test code is in this [repo](https://github.com/hopezh/test_rhino3dm_js/blob/main/index.html). 

(BTW, I init the test repo with `npm create vite@latest`.)

# Error

```shell
npm run dev

> test_rhino3dm_js@0.0.0 dev
> vite

  VITE v5.2.11  ready in 217 ms

  ➜  Local:   http://localhost:5173/
  ➜  Network: use --host to expose
  ➜  press h + enter to show help
✘ [ERROR] Could not resolve "ws"

    node_modules/rhino3dm/rhino3dm.js:9:64363:
      9 │ ...r(26)},listen:function(sock,backlog){if(!ENVIRONMENT_IS_NODE){throw new FS.ErrnoError(138)}if(sock.server){throw new FS.ErrnoError(28)}var WebSocketServer=require("ws").Server;var host=sock.saddr;sock.server=new WebSocketServer({host:host,port:sock.sport});Module["websocket"].emit("listen",sock.stream.fd);sock.server.on("connec...
        ╵                                                                                                                                                                       ~~~~

  You can mark the path "ws" as external to exclude it from the bundle, which will remove this error
  and leave the unresolved path in the bundle. You can also surround this "require" call with a
  try/catch block to handle this failure at run-time instead of bundle-time.
```

# my `npm list`:

```shell
> npm list

test_rhino3dm_js@0.0.0 C:\Users\jizha\Downloads\code\test_rhino3dm_js
+-- rhino3dm@8.6.1
`-- vite@5.2.11
```

fraguada, 2024-05-13:
Unfortunately I don't know anything about vite, so don't know how to help. I'll try running vite to see if I can figure out the issue.

jdev-chicago, 2024-09-04:
any updates on this issue? I'm running into the same error. vite@latest : Typescript+SWC

mr07s, 2024-11-22:
I was facing the same issue, but after upgrading to `ws` version 8.18.0, the issue was resolved.

fraguada, 2025-02-28:
I was finally able to try out vite and the only way I suceeded to import rhino3dm is to have the library imported via a script tag in the html. Sure there are ways to use npm and import it as a module, but I am missing something related to how vite deals with dependencies.

I uploaded an example here: https://github.com/mcneel/rhino-developer-samples/tree/8/rhino3dm/js/vite_example

If someone knows how to use `npm i rhino3dm` and then `import rhino3dm from 'rhino3dm'`, I'd like to know!

closed 2025-02-28 by fraguada

fraguada, 2025-03-03:
I see this is a general issue with webpack and handling all webassembly cases

- [Webpack doesn’t work well with wasm modules created with Emscripten · Issue #7352 · webpack/webpack · GitHub](https://github.com/webpack/webpack/issues/7352)
- [Can't find a way to disable WebAssembly parser · Issue #7264 · webpack/webpack · GitHub](https://github.com/webpack/webpack/issues/7264#issuecomment-388112273)

fraguada, 2025-03-03:
Ok. After some digging I update the [vite example](https://github.com/mcneel/rhino-developer-samples/tree/8/rhino3dm/js/vite_example) with a few changes:

1. Import rhino3dm: npm i rhino3dm
2. Change the 'main' entry file to rhino3dm to use the module. This change is in node_modules/rhino3dm/package.json: "main": "rhino3dm.module.js"
3. Create a vite.config.js file in the root of the project with the following contents: