mcneel/rhino3dm #622: Consider building separate python wheels for macOS arm64 & x86-64

robbievanleeuwen, 2024-06-13:
The macOS wheels currently uploaded to PyPI are as follows:

- Python 3.11 & 3.12: `macos-14` (universal), `macos-12` (universal), `macos-11` (universal)
- Python 3.10 and below: `macos-14` (universal), `macos-12` (x86-64), `macos-11` (x86-64)

I've run into a few issues where the universal wheel does not work for the given architecture, e.g. below installed on `macos-14-arm64` (a github actions runner) gives an error saying it has the `x86` version but needs the `arm64` version:

```shell
import sys
    if sys.version_info.major==2:
        from _rhino3dm import *
    else:
>       from ._rhino3dm import *
E       ImportError: dlopen(/Users/runner/work/section-properties/section-properties/.nox/tests-3-12/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so, 0x0002): 
tried: '/Users/runner/work/section-properties/section-properties/.nox/tests-3-12/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')),
'/System/Volumes/Preboot/Cryptexes/OS/Users/runner/work/section-properties/section-properties/.nox/tests-3-12/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(no such file), '/Users/runner/work/section-properties/section-properties/.nox/tests-3-12/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))

.nox/tests-3-12/lib/python3.12/site-packages/rhino3dm/__init__.py:5: ImportError
```

The offending workflow can be found [here](https://github.com/robbievanleeuwen/section-properties/actions/runs/9492534534/job/26159811824) for more info.

Further on my own mac (intel - `x86`) I run into a similar issue where after running a `pip install rhino3dm` in a fresh python 3.12 environment I get an error saying 'have `arm64`, need `x86`':

```shell
Python 3.12.4 (main, Jun 11 2024, 11:59:07) [Clang 15.0.0 (clang-1500.3.9.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import rhino3dm
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/robbie/.pyenv/versions/test/lib/python3.12/site-packages/rhino3dm/__init__.py", line 5, in <module>
    from ._rhino3dm import *
ImportError: dlopen(/Users/robbie/.pyenv/versions/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so, 0x0002): 
tried: '/Users/robbie/.pyenv/versions/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so' 
(mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64h' or 'x86_64')),
'/System/Volumes/Preboot/Cryptexes/OS/Users/robbie/.pyenv/versions/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(no such file), '/Users/robbie/.pyenv/versions/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64h' or 'x86_64')),
'/Users/robbie/.pyenv/versions/3.12.4/envs/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64h' or 'x86_64')),
'/System/Volumes/Preboot/Cryptexes/OS/Users/robbie/.pyenv/versions/3.12.4/envs/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(no such file), '/Users/robbie/.pyenv/versions/3.12.4/envs/test/lib/python3.12/site-packages/rhino3dm/_rhino3dm.cpython-312-darwin.so'
(mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64h' or 'x86_64'))
```

I'm wondering if you would consider splitting the wheels to have discreet `arm64` and `x86` wheels on PyPI to avoid this issue?

Many thanks,
Robbie

fraguada, 2024-06-14:
It could be that we just aren't building the wheels correctly. @jesterKing recently made some changes to the script that we use to build these. Currently it is in dev and will be merged to main with the next release. Hopefully this fixes your issues, and if not, we'll continue working on it.

fraguada, 2024-07-12:
@robbievanleeuwen I've just published rhino3dm 8.9.0-beta with the changes I discussed. I would appreciate if you could test this version and see if it solves your issue related to x86_64 / arm64 wheels on macOS.

robbievanleeuwen, 2024-10-07:
Sorry for the delay in getting back to you - yes v8.9.0 has fixed this!

closed 2024-10-07 by robbievanleeuwen