mcneel/rhino3dm #623: Python: Polyline.GetSegments() returns last element with <NULL> causing segfault

boblyx, 2024-06-13:
Tested on `8.0.1`, `8.4.0`, `8.6.0`, `8.6.1`, on Debian Bookworm, Python 3.10.13 (gcc 12.2)
Given this sample code:
```python
from rhino3dm import Point3d, Polyline
pl = Polyline(4)
points = [[1,1,1], [2,2,2], [3,3,3], [4,4,4]]
for i,p in enumerate(points):
    pl.Add(p[0], p[1], p[2])
a = None
try:
    a = pl.GetSegments()
except Exception as e:
    print(e)
print(a)
```
The print output:
```
(<rhino3dm._rhino3dm.LineCurve object at 0x7ec362f999b0>, <rhino3dm._rhino3dm.LineCurve object at 0x7ec34c143ab0>, <NULL>)
```
Accessing the <NULL> object results in a segmentation fault.
Testing the same thing in RhinoCommon gives me 4 lines as expected.

fraguada, 2024-06-17:
Thanks for reporting this. We were incorrectly creating an array with `Polyline.Count ` instead of `Polyline.Count - 1`. This set of actions should produce some wheels with the fixed code: https://github.com/mcneel/rhino3dm/actions/runs/9544446906

fraguada, 2024-06-17:
@boblyx if an ubuntu build works for you, try: https://github.com/mcneel/rhino3dm/actions/runs/9544446906/artifacts/1607659788

boblyx, 2024-06-18:
Thanks @fraguada I have tested it and the wheel you linked (https://github.com/mcneel/rhino3dm/actions/runs/9544446906/artifacts/1607659788) now displays the correct behaviour.

closed 2024-07-05 by fraguada