mcneel/rhino3dm #637: Fail Brep.CreateFromMesh()

Cool-key, 2024-09-06:
code

[image]
[points.txt](https://github.com/user-attachments/files/16904827/points.txt)

mesh.Normals.ComputeNormals();
mesh.Compact();
string str = SolidCoreLib.Storage_Method.ObjectToJson(mesh.Faces.Select(r=>new List<double[]> { new double[] { mesh.Vertices.ElementAt(r.A).X, mesh.Vertices.ElementAt(r.A).Y, mesh.Vertices.ElementAt(r.A).Z }, new double[] { mesh.Vertices.ElementAt(r.B).X, mesh.Vertices.ElementAt(r.B).Y, mesh.Vertices.ElementAt(r.B).Z }, new double[] { mesh.Vertices.ElementAt(r.C).X, mesh.Vertices.ElementAt(r.C).Y, mesh.Vertices.ElementAt(r.C).Z } }).ToList());
Brep brep =Brep.CreateFromMesh(mesh,false);

brep==NULL

sbaer, 2024-09-06:
Is this a bug report?

closed 2024-09-06 by sbaer

Cool-key, 2024-09-07:
This is not a bug report，points file make up the vertices of the mesh，Running  code ，return brep is null.
mesh.Normals.ComputeNormals();
mesh.Compact();
Brep brep =Brep.CreateFromMesh(mesh,false);

Cool-key, 2024-09-07:
Mesh cannot be converted to brep