mcneel/rhino3dm #638: Missing mesh bindings in rhino3dm.js

prideout, 2024-09-28:
`MeshingParameters` appears in the typescript definitions file, but isn't used anywhere.

Some context: I am currently exporting breps to 3dm from my application, but they appear as wireframes when viewed in McNeel's Web Viewer.  To fix this, my understanding is that I need to "attach" a mesh to the brep, but I don't see how to generate a mesh object from my brep object.

prideout, 2024-09-28:
Ah, when examining the mesh in Rhino with "what", it says "closed polysurface" and it looks fine when I change the View to Shaded.  So maybe this is not really a problem.  :)

fraguada, 2024-09-30:
Hello! It looks like you found one of the main differences between what one can do with Rhino vs what we offer in rhino3dm. Things like meshing breps (your closed polysurface, polysurfaces, etc),  surfaces, and extrusions have to happen with Rhino as we do not offer any sort of brep meshing in rhino3dm. When you turn on some sort of shaded mode in Rhino, the breps get a "Render Mesh" and when you save the file, this render mesh is usually saved as well. 
I have recently exposed "SetMesh" for breps and extrusions so you can provide your own mesh. Here are some very simple examples with spheres and boxes: [[cs](https://github.com/mcneel/rhino-developer-samples/blob/8/rhino3dm/cs/SampleCSTestSetMesh/Program.cs#L14)|[js](https://github.com/mcneel/rhino-developer-samples/blob/8/rhino3dm/js/SampleJSMakeBrep/script.js#L135)|[py](https://github.com/mcneel/rhino-developer-samples/blob/8/rhino3dm/py/SampleBrepSetMesh.py#L12)] 
Of course, this is not trivial for more complex shapes. 
Let us know if you have any more questions or run into anything that could use a tuneup!

closed 2024-09-30 by fraguada