mcneel/rhino3dm #648: TextureMapping for non-Meshes

cedemax, 2024-10-17:
It seems that there is no way to assign texture mapping to non-Meshes like NURBS or Polysurfaces. It is possible to assign a material to these, but not texture mapping. 

Is this something that could be added?

fraguada, 2024-11-27:
@cedemax are you using js, py, or dotnet?nCan you provide an example of something that isn't working?

cedemax, 2024-11-27:
We are using dotnet. More detail:

```c#
// We can add UVs like this to a mesh
var rMesh = new Rhino.Geometry.Mesh();
....
rMesh.TextureCoordinates.Add(...);

// but for Breps there seems to be nothing

var brep = new Rhino.Geometry.Brep();
....

brep.TextureMapping ?? 
brepFace.TextureCoordinates ??
brepTrimCurve.TextureCoordinates ??

// There are things like:
var texMap = TextureMapping.CreatePlaneMapping(...);
// but these can't seem to be applied to Breps.

```

fraguada, 2024-11-27:
Thanks. I see the method we use for Rhino, I see we don't have an analog for rhino3dm. At first glance I don't see anything that would stop us from adding this in rhino3dm. Let me check with the team.

cedemax, 2025-01-24:
Any update on this?