mcneel/rhino3dm #664: [FEATURE REQUEST] AddPoint with attributes

StudioWEngineers, 2024-12-14:
I need to add points to a 3dm file with attributes via Python, primarily to assign them to a specific layer.

The two ways to do it I see are:

1. add another argument to the constructor `AddPoint1`, defaulting it to `nullptr` to keep the current behaviour.
https://github.com/mcneel/rhino3dm/blob/2be29689d99a9b8c0b511e3646cbe38a0bad6a81/src/bindings/bnd_extensions.cpp#L422
3. update and reactivate the methods `AddPoint3` and `Addpoint4` .
https://github.com/mcneel/rhino3dm/blob/2be29689d99a9b8c0b511e3646cbe38a0bad6a81/src/bindings/bnd_extensions.h#L33

I implemented 1) in my fork and it seems to work well, but it would be nice to have it in the upstream.
Would a PR in principle be accepted?

Also related to
https://discourse.mcneel.com/t/adding-attributes-to-a-rhino3dm-point3d/153992

fraguada, 2024-12-14:
I would go with adding `AddPoint3` and `AddPoint5` which both take object attributes.

StudioWEngineers, 2024-12-15:
OK!
Thank you for the swift reply!

fraguada, 2025-01-08:
Thanks for your very useful and thorough contributions!

closed 2025-01-08 by fraguada